talkkonnect/talkkonnect #142: ALSA Config errors

MattWanjia, 2025-02-20:
I am using the talkkonnect image for raspi 4. by default, my speaker works fine and the mic does not transmit. As soon as i override the alsa config as below
"""
pcm.!default {
    type asym
    playback.pcm "plughw_output"
    capture.pcm "hw_input"
}

pcm.hw_input {
    type hw
    card 0
    device 0
}

pcm.plughw_output {
    type plug
    slave.pcm "plughw:0,0"
}

ctl.!default {
    type hw
    card 0
}
"""
my mic works but the speakjer stops working. i have done all the tests with alsa, tx and rx simultaneously and the speaker and mic work fine.

arecord -D hw:0,0 -f S16_LE -r 48000 -c 1 | aplay -D plughw:0,0

talkkonnect, 2025-02-21:
I have been using cm108 USB sound card for both mic and speaker. Also have experience with seperate usb mic and usb speaker all I have had to do is config /usr/share/alsa/alsa.conf and change the default device. 
You have provided no information on what sound card you use or what is your setup, really don't know what to suggest if there is no information except the config file.

talkkonnect, 2025-03-04:
since no update for 2 weeks i will close this ticket

closed 2025-03-04 by talkkonnect