talkkonnect/talkkonnect #143: Systemd service might be conflicting with screen? (Questions)

SecareLupus, 2025-02-20:
Hello, I'm trying to run talkkonnect at boot, and I had initially tried to do so using the systemd method listed in [this blog post](https://www.talkkonnect.com/2020/06/15/various-methods-for-starting-talkkonnect/)

My devices don't want to stay connected, and the logs seemed to indicate they were being killed as part of a restart process. Is it possible that systemd isn't able to keep track of the execution status of the screen session, and restarting them/disabling them because it thinks it's crashing?

Can talkkonnect be launched without screen safely? I don't believe we access the screen session in our current system design, so it's probably fine for our use case if it simply runs as a daemon.

Is there a better recommended way to do this at this point? I know the blog post I'm following is almost 5 years old at this point.

Thanks for your help!

talkkonnect, 2025-02-21:
use the script in github in conf/system.d you dont need to connect to the screen as talkkonnect also supports ssh directly into talkkonnect see and configure the xml file to do this

suvir

talkkonnect, 2025-03-04:
since no update for 2 weeks i will close this ticket

closed 2025-03-04 by talkkonnect