talkkonnect/talkkonnect #144: Systemd

SecareLupus, 2025-03-06:
This is a followup to #143, which I'm unable to re-open.

I've switched to the systemd file in the project folders, but it doesn't seem to have made a difference. The Talkkonnect systemd services on several devices have stopped and disabled themselves for reasons I can't seem to identify. Output from `journalctl -u talkkonnect -b` doesn't look like much to me, except that the usage information for the kill command looks to me like maybe the restart or stop commands are failing to execute due to a syntax error?

```
Mar 06 07:19:00 sgl-eb-0369 systemd[1]: Starting talkkonnect.service - TalKKonnect Radio Screen...
Mar 06 07:19:00 sgl-eb-0369 systemd[1]: Started talkkonnect.service - TalKKonnect Radio Screen.
Mar 06 07:19:30 sgl-eb-0369 kill[1567]: Usage:
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  kill [options] <pid> [...]
Mar 06 07:19:30 sgl-eb-0369 kill[1567]: Options:
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  <pid> [...]            send signal to every <pid> listed
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -<signal>, -s, --signal <signal>
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:                         specify the <signal> to be sent
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -q, --queue <value>    integer value to be sent with the signal
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -l, --list=[<signal>]  list all signal names, or convert one to a name
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -L, --table            list all signal names in a nice table
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -h, --help     display this help and exit
Mar 06 07:19:30 sgl-eb-0369 kill[1567]:  -V, --version  output version information and exit
Mar 06 07:19:30 sgl-eb-0369 kill[1567]: For more details see kill(1).
Mar 06 07:19:30 sgl-eb-0369 systemd[1]: talkkonnect.service: Control process exited, code=exited, status=1/FAILURE
Mar 06 07:19:30 sgl-eb-0369 systemd[1]: talkkonnect.service: Failed with result 'exit-code'.
Mar 06 07:19:41 sgl-eb-0369 systemd[1]: talkkonnect.service: Scheduled restart job, restart counter is at 1.
Mar 06 07:19:41 sgl-eb-0369 systemd[1]: Stopped talkkonnect.service - TalKKonnect Radio Screen.
Mar 06 07:19:41 sgl-eb-0369 systemd[1]: Starting talkkonnect.service - TalKKonnect Radio Screen...
Mar 06 07:19:41 sgl-eb-0369 systemd[1]: Started talkkonnect.service - TalKKonnect Radio Screen.
Mar 06 07:20:11 sgl-eb-0369 kill[1852]: Usage:
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  kill [options] <pid> [...]
Mar 06 07:20:11 sgl-eb-0369 kill[1852]: Options:
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  <pid> [...]            send signal to every <pid> listed
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -<signal>, -s, --signal <signal>
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:                         specify the <signal> to be sent
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -q, --queue <value>    integer value to be sent with the signal
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -l, --list=[<signal>]  list all signal names, or convert one to a name
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -L, --table            list all signal names in a nice table
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -h, --help     display this help and exit
Mar 06 07:20:11 sgl-eb-0369 kill[1852]:  -V, --version  output version information and exit
Mar 06 07:20:11 sgl-eb-0369 kill[1852]: For more details see kill(1).
```

This is the output from `systemctl status talkkonnect` (this output is from another device showing the same symptoms as the previous journalctl output, so the times and hostname might not line up, but the issue is otherwise exactly the same)
```
× talkkonnect.service - TalKKonnect Radio Screen
     Loaded: loaded (/etc/systemd/system/talkkonnect.service; disabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2025-03-06 07:24:12 EST; 6h ago
   Duration: 2min 56.511s
    Process: 2423 ExecStop=/bin/kill -s SIGTERM $MAINPID (code=exited, status=0/SUCCESS)
   Main PID: 1053 (code=exited, status=1/FAILURE)
        CPU: 1.474s

Mar 06 07:21:15 sgl-eb-0370 systemd[1]: Starting talkkonnect.service - TalKKonnect Radio Screen...
Mar 06 07:21:15 sgl-eb-0370 systemd[1]: Started talkkonnect.service - TalKKonnect Radio Screen.
Mar 06 07:24:12 sgl-eb-0370 systemd[1]: Stopping talkkonnect.service - TalKKonnect Radio Screen...
Mar 06 07:24:12 sgl-eb-0370 systemd[1]: talkkonnect.service: Main process exited, code=exited, status=1/FAILURE
Mar 06 07:24:12 sgl-eb-0370 systemd[1]: talkkonnect.service: Failed with result 'exit-code'.
Mar 06 07:24:12 sgl-eb-0370 systemd[1]: Stopped talkkonnect.service - TalKKonnect Radio Screen.
Mar 06 07:24:12 sgl-eb-0370 systemd[1]: talkkonnect.service: Consumed 1.474s CPU time.
```

Device is a Raspberry Pi 4B running a custom build of Debian 12. I'm not sure how to get more information on what is causing the service to occasionally shuts down, or why it's disabling the service entirely when it does.

talkkonnect, 2025-03-07:
I have not seen this problem before, if you can allow ssh to that device i can remotely check it for you.

SecareLupus, 2025-03-10:
I've sent an email to the address on your github account with access links to three of our devices.

talkkonnect, 2025-03-13:
Does the problem happen on raspbian with the image? All I see is a more detailed systemd start up with private tmp directory that is different.

SecareLupus, 2025-03-17:
As in stock Raspberry Pi OS? I don't know, we don't have vanilla devices in production because we're adding the application as a feature to an existing device. I can spin up a Raspberry Pi 4B with stock Pi OS, and install talkkonnect on it, but as the issue seems to happen unpredictably, I can't guarantee that it will trigger. It would be great if there was a log entry with why the application closed so we could look at the log and maybe backtrace the issue from there.

Our current system is designed to use the latest installation of talkkonnect without alterations, except to the configuration XML file and a few different attempts at ways to configure the systemd service file. There are modifications to ALSA configuration, providing concrete bindings to connect talkkonnect to our particular USB dongle.

SecareLupus, 2025-03-17:
I should note, we *are* using stock Raspberry Pi OS 64bit for our base, we just have some internal customisations which get layered on top at install or first boot. For example, talkkonnect gets installed on firstboot.

talkkonnect, 2025-03-18:
From ssh I can only see changes in the systemd script and also different is from what I use. 

From the little time I have available and looked I don’t know what is causing this intermittent behavior on your side.

Have not faced the issue you are facing, also other people who use talkkonnect in production also have not reported such issues so I am at this moment in the dark.

SecareLupus, 2025-03-21:
I understand, I don't think it's your responsibility to trace down this weird corner case I've probably created for myself, but is there any way that talkkonnect can log somewhere what it's reason for quitting is, whether instructed to, or because of an error state? If I had some indication of why it's quitting out, or why it's failing to reconnect afterwards, that might provide the breadcrumbs to figure out the cause of the issue.

talkkonnect, 2025-03-21:
3 things come to mind
1. In systemd remove private tmp
2. there is a check called singleinstance configurable from xml that kills talkkonnect if it detects a second running instance, try to set singleinstance to false in xml and see
3. you can do logging to file set the logging to screenandfilewithlineno 

That’s all I can think of at the moment

Suvir