talkkonnect/talkkonnect #146: Update script troubles

NizamusRuslan, 2025-03-18:
After installing prebuild image to orange pi zero, i try to update by  https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect.sh ...
go: go.mod file not found in current directory or any parent directory.
        'go get' is no longer supported outside a module.
        To build and install a command, use 'go install' with a version,
        like 'go install example.com/cmd@latest'
        For more information, see https://golang.org/doc/go-get-install-deprecation
        or run 'go help get' or 'go help install'.
./update-talkkonnect.sh: line 106: cd: /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect: No such file or directory
cannot find package "cmd/talkkonnect/main.go" in any of:
        /usr/local/go/src/cmd/talkkonnect/main.go (from $GOROOT)
        /home/talkkonnect/gocode/src/cmd/talkkonnect/main.go (from $GOPATH)
cp: cannot stat '/root/mumble-20250318-150342.pem': No such file or directory
cp: cannot create regular file '/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/talkkonnect.xml': No such file or directory
talkkonnect is stopped now restarting talkkonnect
Job for talkkonnect.service failed because the control process exited with error code.
See "systemctl status talkkonnect.service" and "journalctl -xeu talkkonnect.service" for details.
=> Finished Updating TalKKonnect
=> Updated talkkonnect binary is in /home/talkkonect/bin
copied old talkkonnect.xml file and replaced in /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/
Happy talkkonnecting!!

Thank you!

talkkonnect, 2025-03-18:
Yes the update breaks the installation as the newer versions of go and also os cause issues. This is a known issue. I will have to build a new image with the latest os and go version and release the image again as the image available on the net is a year old now and many underlying changes have taken place.

So you can also start clean from stock operating system and use to build.sh script to install the latest version.

talkkonnect, 2025-06-01:
Copy and save your talkkonnect.xml file and remove the talkkonnect directory and stop using this update srcipt in the old version. instead use the script tk-build-v1.sh to build from scratch if you are using the image. Allow me to close this ticket. I will release new images time permitting,

closed 2025-06-01 by talkkonnect