talkkonnect/talkkonnect #148: talKKonnect builds in PC's and virtual machines and making cgo and gopus work again

zrdimetc, 2025-04-20:
This is in relation to old issues [74](https://github.com/talkkonnect/talkkonnect/issues/74) and [99](https://github.com/talkkonnect/talkkonnect/issues/99).

This is a working version of [config header](https://gist.github.com/zrdimetc/e7d9e77a076d8e41f3a124967c1131b8) file for [gopus](https://github.com/layeh/gopus) (Go binding for Opus used with talKKonnect).

Opus config header file `config.h` can make a difference for a successful talKKonnect build in PC's.

If x86/64 build was unsucessful, 1. replace the config header file in libopus development folder in gopus module, 2. `go clean --cache` 3. build talkkonnect again.

I updated the gopus with the latest Opus version 1.5.2 from Apr 12, 2024. gopus with Opus codec 1.5.2 is [here](https://github.com/zrdimetc/gopus152). To build x86/64 talkkonnect with that version of gopus, just replace old gopus module files with the new files, clean go cache and recompile.

Older versions of gopus with Opus 1.3.1 and 1.1.2 are [here](https://github.com/zrdimetc/gopus131) and [here](https://github.com/zrdimetc/gopus112). They are also working with x86/64 talkkonnect.

Opus version 1.5 delivered some [major changes](https://opus-codec.org/demo/opus-1.5/) including,
- significant improvements to packet loss robustness using Deep Redundancy (DRED).
- Packet loss concealment through Deep PLC.
- Low-bitrate speech quality enhancement down to 6 kb/s wideband.
- Improved x86 (AVX2) and Arm (Neon) optimizations.
- Support for 4th and 5th order ambisonics.

talKKonnect PC builds. if taking a libopus 1.5.2 for Go bindings and non-shared libraries will benefit from the most recent Opus codec 1.5 improvements. Raspberry Pi builds work with shared libraries from libopus-dev package, and in current 2024/2025 Linux distros they are Opus 1.3.1.

talkkonnect, 2025-04-21:
Zoran,

Can we use this new library with the raspberry pi? have you tested it? does it work well.

thanks for this contribution,

suvir

zrdimetc, 2025-04-21:
> Zoran,
> 
> Can we use this new library with the raspberry pi? have you tested it? does it work well.
> 
> thanks for this contribution,
> 
> suvir

Hi Suvir. Yes. gopus files from the provided links can be safely used with Raspberry Pi's, Raspberry Pi build process only use the shared libraries from `libopus-dev` OS package. Currently, the shared libraries will be from Opus version 1.3.1. For Raspberry Pi, only the file `opus_shared.go` file is required to build sucessfully. Even if the entire `opus_1.x.x` folder with C programs and headers from gopus is removed together with `opus_nonshared.go` file, the Raspberry Pi talKKonnects will compile fine, since they dont care about static nonshared libraries. But the PC's and vm's, amd64/i386 are another story. They need the `opus_nonshared.go` and Opus source code files in the `gopus` module. The most critical file for buidling success is config header `config.h` file inside the Opus files folder and it is the file that needs careful review and adjustment for PC build process to work. I did that, while testing the PC builds. 

For fixing the PC builds, I suggest 2 changes to be made.

1. Please take the most recent [`config.h`](https://gist.github.com/zrdimetc/e7d9e77a076d8e41f3a124967c1131b8) to replace old `config.h` file from the currently supplied gopus [gopus](https://github.com/talkkonnect/gopus) package by talkkonnect. This will fix the current version of talkkonnect provided gopus module for building in PC's with Opus version 1.3.1. The new config header had been tested to work. I built talKKonnect in amd64 physical machines and vm's with it. This change does not affect Raspberry Pi builds at all. This config will work with any Opus, 1.52, 1.3.1, 1.1.2.
 
2. It's time to upgrade old Opus 1.3.1 files in the gopus with new Opus 1.5.2 files. You can take new version of gopus from [here](https://github.com/zrdimetc/gopus152). `opus_shared.go` file is unchanged. But please note, [opus_nonshared.go](https://github.com/zrdimetc/gopus152/blob/main/opus_nonshared.go) is new to make it work with Opus 1.5.2.

Another way to update gopus to 1.5.2 is to take [libopus files](https://opus-codec.org/release/stable/2024/04/12/libopus-1_5_2.html) directly from Opus project. Replace the old [opus_nonshared.go](https://github.com/talkkonnect/gopus/blob/master/opus_nonshared.go) with the [new one](https://github.com/zrdimetc/gopus152/blob/main/opus_nonshared.go). Leave `opus_shared.go` unchanged. There is a change in the new nonshared file for new Opus. Take the provided new config.h from my link. 

Hopefully, with the proposed changes, the talKKonnect PC builds will work smooth again and with the benefit of new Opus codec v.1.5.2.

talkkonnect, 2025-04-23:
> To use new Opus 1.5.2 with talKKonnect in Raspberry Pi (ARMv8), the easiest way would be -
>
>   1.  First check your installed Opus version with opus-tools, opusenc -v.
> It should return
>
> opusenc -V
> opusenc opus-tools 0.2 (using libopus 1.3.1)
> Copyright (C) 2008-2018 Xiph.Org Foundation
>
>   1.  Purge libopus 1.3.1 package
> apt-get purge libopus0 libopus-dev
>
>   2.  Get Opus codec 1.5.2 runtime library from http://ftp.us.debian.org/debian/pool/main/o/opus/libopus0_1.5.2-2_arm64.deb
>
>   3.  Get Opus codec 1.5.2 library development files from http://ftp.us.debian.org/debian/pool/main/o/opus/libopus-dev_1.5.2-2_arm64.debFor older 32 bit boards get appropriate .deb "sid" packages fromhttps://packages.debian.org/sid/libopus0<https://packages.debian.org/sid/libopus-dev> and
> https://packages.debian.org/sid/libopus0
>
>   4.  Install Opus 1.5.2 packagesdpkg -i libopus0_1.5.2-2_arm64.deb
> dpkg -i libopus-dev_1.5.2-2_arm64.deb
>
>   5.  Check Opus version
>
> opusenc -V
> opusenc opus-tools 0.2 (using libopus 1.5.2)
> Copyright (C) 2008-2018 Xiph.Org Foundation
> [... 37 lines cut ...]
> happily with new Opus.
>
>   2.  This is not the cleanest method of upgrading. If you need to revert back and uninstal Opus 1.5.2 you need to make up your mind about what works for you to go back without breaking the system? You could take old Opus 1.3.1 libraries from the "Bookworm" repository again and reinstall them, replacing 1.5.2.
>
>   3.  In terms of stability and CPU utilization, I didn't see anything out of the ordinary with the new Opus 1.5.2 compared to 1.3.1. How much talkkonnect can benefit from the promissing new 1.5 features it's too early to say. So far so good, "the transplant" of new codec was successful.

Great work as usual Zoran thanks for your experimentation and contributions

Suvir

zrdimetc, 2025-04-27:
Thank you Suvir. Until the gopus `config.h` file can be replaced or updated in the official talkkonnect/gopus module, here is quick fix for a PC build - to temporarily switch an old talkkonnect/gopus module with the new one.

1.  Try to build talkkonnect as usual.

2. After the build fails for PC, do`echo "replace github.com/talkkonnect/gopus v0.0.0-20220213040034-763f05853b2f => github.com/zrdimetc/gopus152 v0.0.0-20250419171437-6bec3b0dbec9" >> /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/go.mod
`
This will amend the `go.mod` from the official talkkonnect repository, to temporarily work with the new gopus module with Opus 1.5.2.

3. Build talkkonnect again. It will build successfully in a PC this time.

talkkonnect, 2025-06-01:
uploaded the file tk-build-v1.sh as sent by zoran

closed 2025-06-01 by talkkonnect