talkkonnect/talkkonnect #150: Sample xml config to reflect the new http api volume commands

zrdimetc, 2025-05-24:
After introducing a new set of http api commands in Feburary last year, for volume control to differentiate between input and output device volume level, or figuratively speaking "tx" and "rx", old talkkonnect xml config examples need updating.

Old http api commands, 

```
currentvolume
volumeup
volumedown

```
are now obsolete. 

The new http api commands are:

```
currentrxvolume
volumerxup
volumerxdown
volumetxup
volumetxdown
currenttxvolume

```
Unless the `talkkonnect.xml` is updated to reflect this change, http api volume controls won't work. 
To address the issue, find old http volume api command entries in production `talkkonnect.xml` files,

```
<!-- <command action="currentvolume" funcparamname="" message="Current Volume" enabled="true"/> -->
<!-- <command action="volumeup" funcparamname="" message="Volume Up" enabled="true"/> -->
<!-- <command action="volumedown" funcparamname="" message="Volume Down" enabled="true"/> -->
```

and “sanitize” them.

Add the new http api commands, under,
```
<remotecontrol>
  <http listenport="8080" enabled="true">
  ...
  </http>

```
config file section.

e.g.

```
<command action="currentrxvolume" funcparamname="" message="Current RX Volume" enabled="true"/>
<command action="volumerxup" funcparamname="" message="Volume RX Up" enabled="true"/>
<command action="volumerxdown" funcparamname="" message="Volume RX Down" enabled="true"/>
<command action="currenttxvolume" funcparamname="" message="Current TX Volume" enabled="true"/>
<command action="volumetxup" funcparamname="" message="Volume TX Up" enabled="true"/>
<command action="volumetxdown" funcparamname="" message="Volume TX Down" enabled="true"/>
```

After fixing the config file, http api volume control should work again.

zrdimetc, 2025-05-25:
Same goes for mqtt api volume commands. Instead of 3 old commands, there are 6 new commands now and this needs to be reflected in the xml config file, for mqtt volume control to work.

Old:
```
<!--<command action="currentvolume" message="Current Volume" enabled="true"/>
<command action="volumeup" message="Volume Up" enabled="true"/>
<command action="volumedown" message="Volume Down" enabled="true"/>-->

```
New:
```
<command action="currentrxvolume" message="Current Rx Volume" enabled="true"/>
<command action="volumerxup" message="Volume Rx Up" enabled="true"/>
<command action="volumerxdown" message="Volume Rx Down" enabled="true"/>
<command action="currenttxvolume" message="Current Tx Volume" enabled="true"/>
<command action="volumetxup" message="Volume Tx Up" enabled="true"/>
<command action="volumetxdown" message="Volume Tx Down" enabled="true"/>

```

zrdimetc, 2025-05-31:
As pointed out by Suvir in the sample-configs readme, only the `talkkonnect.tkv1pcb` config sample is always getting updated with latest xmlparser changes. While the other sample-configs files remain unchanged. That is a starting point. Therefore, only the `talkkonnect.tkv1pcb` should be downloaded to talkkonnect program directory and renamed `talkkonnect.xml` . Older configs are kept in the sample-configs folder for historical reasons. One very important new option in the config is offest by 512 for the new numbering scheme in Raspbian system.
<hardware targetboard="rpi">
      <gpiooffset>512</gpiooffset>
      ...
</hardware>
If in doubt whether the xml config is current with latest supported features, it is always a good idea to read through the latest `xmlparser.go` and compare it with `talkkonnect.xml`.
The issue can be closed. This wasn't a real issue, just an observation.

talkkonnect, 2025-06-01:
Ok I have fixed the file and will upload as talkkonnect.current-sample-config.xml

closed 2025-06-01 by talkkonnect