talkkonnect/talkkonnect #156: sub channel

gonecrazy25, 2025-07-25:
I am trying to have the takkonnect server join a sub-channel but I don't know what type of format I should use.  Also, if I want it to join a private channel that has a password how would you put that in the XML file?

talkkonnect, 2025-07-26:
i dont think i got the sub-channel thing working yet it was all to confusing now can join main channel. when i figure it out i will let you know

talkkonnect, 2025-07-28:
I found and fixed the issue now you have to use the latest release of talkkonnect for this to work. Version 2.44.01 released 28 July 2025.

for example in the channel xml tag to connect to a sub-channel called test which is a subchannel of the channel HAM-CB the xml tag would look like this <channel>HAM-CB,test</channel>

it works I have tested it and released the code on github.com

suvir

gonecrazy25, 2025-07-29:
I can verify that this works to join a sub-channel now.  The problem I'm having now is joining a password-protected sub-channel or even a regular password-protected channel. I put the password in the Tokens list, but it still says that it is unable to find channel name.  Is the tokens list the correct place to put the password for the channel?

There also seems to be a bug that you might be aware of, if you delete a channel on the Mumble server while talkkonnect is connected, even if it is not in that channel it seems to crash talkkonnect and it restarts.  It's not a big deal, but I was unsure if you were aware of it.

talkkonnect, 2025-07-30:
The tokens can be defined in XML and they were working before this sub-channel change. Let me test it over the weekend and try to reproduce this bug and fix it when I have some time.

As for the removed channel since I keep the channel in an array if the channel is removed and talkkonnect crashes this is expected behavior for the current way the code is implemented. To fix this would require major surgery. I don't think people keep adding and removing channels as a daily thing so I would consider its effect and priority low on the list.

Also if talkkonnect crashes for any reason if using the systemd scripts provided it will start back up itself.

gonecrazy25, 2025-08-04:
I did want to let you know that tokens are working for channels that are single.  They don't seem to work if the channel has sub-channels in it or if it is a sub-channel that you are trying to join.

I would agree that the channel remove crash is very low priority, as talkkonnect does restart with no problem automaticly and like you said, once you have your mumble server channels set very rarely would you need to delete and add them.  I just wanted to be sure you were aware of it.

Thank you for this great product it is very neat how it all works together!