talkkonnect/talkkonnect #157: Talkkonnect on a Debian machine #2

UgoPoddine, 2025-07-30:
Following the Suvir's suggestion in issue 155, I have adopted the first solution (tk-build-v1.sh), but, after having installed all dependecies (including the ones suggested by the Thomas's solution), the compilation process stops here :

/usr/bin/ld: /home/talkkonnect/gocode/pkg/mod/github.com/talkkonnect/[gopus@v0.0.0-20220213040034-763f05853b2f](mailto:gopus@v0.0.0-20220213040034-763f05853b2f)/opus-1.3.1/src/opus_encoder.c:1863: undefined reference to `celt_fatal'
collect2: error: ld returned 1 exit status.

The go version is the last :
go version go1.24.5 linux/amd64

Must I install also celt as dependecy ?

Thank-you for any help

IU1IPB Ugo

talkkonnect, 2025-08-01:
celt is already a part of opus, have to get the OPUS library to compile correctly that's the issue. You will have to use layeh who created bernard library i think and change the references in the source code to get it to compile. Not an easy task since there are many things to do if the tk-build-v1.sh script doesnt work.

UgoPoddine, 2025-08-01:
Thank-you.
At the moment I have libopus0 libopus-dev and opus-tools correctly installed from bookworm repo and mumble client well running, but 
Any suggestion more ?

Thank-you
Ugo

UgoPoddine, 2025-08-01:
I have applied the "contributed by Thomas" solution, and I was able to compile.
Thank-you so much.
My suggestion is to include it in the script, in order to make easyer to use talkonnect on non-pi machines.

Ugo

closed 2025-08-01 by UgoPoddine

zrdimetc, 2025-08-01:
The build script is working. But you have to be careful about Opus config header in gopus module. Check the `go.mod` for gopus version and particularly inspect rigorously Opus `config.h` header. Inside the gopus module from talkkonnect itself `/home/talkkonnect/gocode/pkg/mod/github.com/talkkonnect/gopus@v0.0.0-20220213040034-763f05853b2f` inspect the config.h header? If you pulled a current version with troublesome `config.h` that breaks PC builds, you have to clean module cache `go clean -modcache`.  Or replace bad `config.h` manually and dont download module again, just run `go build`. For a quick fix, you can take gopus 1.3.1. from here [(https://github.com/zrdimetc/gopus131)] or just take a working [config.h](url). Compare `config.h` from that version with `config.h `from here [(https://github.com/talkkonnect/gopus/blob/master/opus-1.3.1/config.h)] and you will see what flags break gopus build in PC? You could also use Gopus with new Opus 1.5.2 from here https://github.com/zrdimetc/gopus152. To solve this issue for good, the `config.h` in talkkonnect's gopus repository needs updating to remove bad config.h arguments that break the PC build. Diff between the two config.h versions show what needs removing in config.h.

zrdimetc, 2025-08-01:
This ia s working config.h for gopus with Opus 1.3.1 that works for PC builds. Neither version of config.h has any issues with Pi builds. Compare for a difference to a config that resides in your system that won't work for PC builds and use the new safe config and you will have solved the build isssue. Just don't download a gopus with bad config.h again. Overwrite a bad config with good config and rebuild. After a config.h from talkkonnect gopus repository is updated, this will no longer be a problem.

```
/* config.h.  Generated from config.h.in by configure.  */
/* config.h.in.  Generated from configure.ac by autoheader.  */

/* Get CPU Info by asm method */
/* #undef CPU_INFO_BY_ASM */

/* Get CPU Info by c method */
/* #undef CPU_INFO_BY_C */

/* Custom modes */
/* #undef CUSTOM_MODES */

/* Do not build the float API */
/* #undef DISABLE_FLOAT_API */

/* Assertions */
/* #undef ENABLE_ASSERTIONS */

/* Debug fixed-point implementation */
/* #undef FIXED_DEBUG */

/* Compile as fixed-point (for machines without a fast enough FPU) */
/* #undef FIXED_POINT */

/* Float approximations */
/* #undef FLOAT_APPROX */

/* Fuzzing */
/* #undef FUZZING */

/* Define to 1 if you have the <alloca.h> header file. */
/* #undef HAVE_ALLOCA_H */

/* Define to 1 if you have the <dlfcn.h> header file. */
#define HAVE_DLFCN_H 1

/* Define to 1 if you have the <inttypes.h> header file. */
#define HAVE_INTTYPES_H 1

/* Define to 1 if you have the `lrint' function. */
#define HAVE_LRINT 1

/* Define to 1 if you have the `lrintf' function. */
#define HAVE_LRINTF 1

/* Define to 1 if you have the <memory.h> header file. */
#define HAVE_MEMORY_H 1

/* Define to 1 if you have the <stdint.h> header file. */
#define HAVE_STDINT_H 1

/* Define to 1 if you have the <stdlib.h> header file. */
#define HAVE_STDLIB_H 1

/* Define to 1 if you have the <strings.h> header file. */
#define HAVE_STRINGS_H 1

/* Define to 1 if you have the <string.h> header file. */
#define HAVE_STRING_H 1

/* Define to 1 if you have the <sys/stat.h> header file. */
#define HAVE_SYS_STAT_H 1

/* Define to 1 if you have the <sys/types.h> header file. */
#define HAVE_SYS_TYPES_H 1

/* Define to 1 if you have the <unistd.h> header file. */
#define HAVE_UNISTD_H 1

/* Define to 1 if you have the `__malloc_hook' function. */
#define HAVE___MALLOC_HOOK 1

/* Define to the sub-directory in which libtool stores uninstalled libraries.  */
#define LT_OBJDIR ".libs/"

/* Make use of ARM asm optimization */
/* #undef OPUS_ARM_ASM */

/* Use generic ARMv4 inline asm optimizations */
/* #undef OPUS_ARM_INLINE_ASM */

/* Use ARMv5E inline asm optimizations */
/* #undef OPUS_ARM_INLINE_EDSP */

/* Use ARMv6 inline asm optimizations */
/* #undef OPUS_ARM_INLINE_MEDIA */

/* Use ARM NEON inline asm optimizations */
/* #undef OPUS_ARM_INLINE_NEON */

/* Define if compiler support EDSP Instructions */
/* #undef OPUS_ARM_MAY_HAVE_EDSP */

/* Define if compiler support MEDIA Instructions */
/* #undef OPUS_ARM_MAY_HAVE_MEDIA */

/* Define if compiler support NEON instructions */
/* #undef OPUS_ARM_MAY_HAVE_NEON */

/* Compiler supports ARMv7 Neon Intrinsics */
/* #undef OPUS_ARM_NEON_INTR */

/* Define if binary requires EDSP instruction support */
/* #undef OPUS_ARM_PRESUME_EDSP */

/* Define if binary requires ARMv6 media instruction support */
/* #undef OPUS_ARM_PRESUME_MEDIA */

/* Define if binary requires NEON instruction support */
/* #undef OPUS_ARM_PRESUME_NEON */

/* This is a build of OPUS */
#define OPUS_BUILD /**/

/* Use run-time CPU capabilities detection */
/* #undef OPUS_HAVE_RTCD */

/* For x86 sse2 instrinsics optimize */
/* #undef OPUS_X86_MAY_HAVE_SSE2 */

/* For x86 sse4.1 instrinsics optimizations */
/* #undef OPUS_X86_MAY_HAVE_SSE4_1 */

/* Define to the address where bug reports for this package should be sent. */
#define PACKAGE_BUGREPORT "opus@xiph.org"

/* Define to the full name of this package. */
#define PACKAGE_NAME "opus"

/* Define to the full name and version of this package. */
#define PACKAGE_STRING "opus 1.3.1"

/* Define to the one symbol short name of this package. */
#define PACKAGE_TARNAME "opus"

/* Define to the home page for this package. */
#define PACKAGE_URL ""

/* Define to the version of this package. */
#define PACKAGE_VERSION "1.3.1"

/* Define to 1 if you have the ANSI C header files. */
#define STDC_HEADERS 1

/* Make use of alloca */
/* #undef USE_ALLOCA */

/* Use C99 variable-size arrays */
#define VAR_ARRAYS 1

/* Define to empty if `const' does not conform to ANSI C. */
/* #undef const */

/* Define to `__inline__' or `__inline' if that's what the C compiler
   calls it, or to nothing if 'inline' is not supported under any name.  */
#ifndef __cplusplus
/* #undef inline */
#endif

/* Define to the equivalent of the C99 'restrict' keyword, or to
   nothing if this is not supported.  Do not define if restrict is
   supported directly.  */
#define restrict __restrict
/* Work around a bug in Sun C++: it does not support _Restrict or
   __restrict__, even though the corresponding Sun C compiler ends up with
   "#define restrict _Restrict" or "#define restrict __restrict__" in the
   previous line.  Perhaps some future version of Sun C++ will work with
   restrict; if so, hopefully it defines __RESTRICT like Sun C does.  */
#if defined __SUNPRO_CC && !defined __RESTRICT
# define _Restrict
# define __restrict__
#endif
```