talkkonnect/talkkonnect #159: Panic (index out of range) on ClientStart with v2 XML - Raspberry Pi 3 (32bit OS)

wainy54, 2025-09-28:
When starting talKKonnect (v2.44.01 Rel 28 July 2025 build) the client successfully reads the config and connects to the Murmur server, but then panics with `index out of range [0] with length 0` in `client.go:566`. The panic appears to be caused either by an empty slice being indexed during channel/account initialization  OR my XML is bad:

[tk-config.xml](https://github.com/user-attachments/files/22583787/tk-config.xml)

This bug reproduces reliably on my Raspberry Pi 3 (using 32bit OS, I'm using this as I plan to deploy on an older Pi later) when using the attached `talkkonnect.xml` (v2-compatible) or older templates I had originally used. I believe the panic occurs because some slice (channels, listen list, or voice target arrays) is not populated but is subsequently indexed without a bounds check.

Here is the trace of the crash:
`panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
github.com/talkkonnect/talkkonnect.(*Talkkonnect).ClientStart(0x2aa0008)
    /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:566 +0x3734
github.com/talkkonnect/talkkonnect.Init({0x7ee9e74d, 0x21}, {0x7a69ac, 0x1})
    /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:235 +0x1654
main.main()
    /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/cmd/talkkonnect/main.go:78 +0x3a4
`

I'm starting talkkonnect like for debug:
sudo -u talkkonnect /home/talkkonnect/bin/talkkonnect -config=/home/talkkonnect/talkkonnect.xml
But I plan to run using a wrapper shell script started from a systemd service.

Here is the output log:
`?1049h22;0;0t?1h?25lH2J0;32m[  info ] 2025/09/28 17:21:21 Successfully Read file talkkonnect.xml
0;32m[	info ] 2025/09/28 17:21:21 Starting XML Configuration Sanity and Logical Checks
0;32m[	info ] 2025/09/28 17:21:21 Finished XML Configuration Sanity and Logical Checks Without Any Alerts/Errors/Warnings
0;32m[	info ] 2025/09/28 17:21:21 Successfully loaded XML configuration file into memory
0;32m[	info ] 2025/09/28 17:21:21 client.go:131:0m Loglevel Set to Debug
0;32m[	info ] 2025/09/28 17:21:21 client.go:183:0m MQTT Server Subscription Disabled in Config
0;32m[	info ] 2025/09/28 17:21:21 client.go:212:0m Connecting to Server 127.0.0.1:64738 Identified As intercom-server With Username intercom-server
0;32m[	info ] 2025/09/28 17:21:21 client.go:250:0m Trying to Open File	 /home/talkkonnect/talkkonnect.log
0;32m[	info ] 2025/09/28 17:21:21 client.go:285:0m [1] Default Mumble Accounts Found in XML config
0;32m[	info ] 2025/09/28 17:21:21 client.go:288:0m Target Board Set as RPI (gpio enabled)
0;96m[ debug ] 2025/09/28 17:21:21 client.go:327:0m Backlight Timer Disabled by Config
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:41:0m 0;44m┌────────────────────────────────────────────────────────────────┐0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:42:0m 0;44m│	_	 _ _	_				_	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:43:0m 0;44m│ | |_ __ _| | | _| | _____  _ __	_ __   ___  ___| |_	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:44:0m 0;44m│ | __/ _` | | |/ / |/ / _ \| '_ \| '_ \ / _ \/ __|  __|	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:45:0m 0;44m│ | || (_| | |   <|   < (_) | | | | | | |	__/ (__| |_	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:46:0m 0;44m│	\__\__,_|_|_|\_\_|\_\___/|_| |_|_| |_|\___|\_ _|\__|	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:47:0m 0;44m├────────────────────────────────────────────────────────────────┤0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:48:0m 0;44m│A Flexible Headless Mumble Transceiver/Gateway for RPi/PC/VM    │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:49:0m 0;44m├────────────────────────────────────────────────────────────────┤0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:50:0m 0;44m│Created By : Suvir Kumar	<suvir@talkkonnect.com>		      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:51:0m 0;44m├────────────────────────────────────────────────────────────────┤0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:52:0m 0;44m│Press the <Del> key for Menu or <Ctrl-c> to Quit talkkonnect    │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:53:0m 0;44m│Additional Modifications Released under MPL 2.0 License	      │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:54:0m 0;44m│Blog at www.talkkonnect.com, source at github.com/talkkonnect   │0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:55:0m 0;44m└────────────────────────────────────────────────────────────────┘0m
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:63:0m Software Ver 2.44.01 Rel 28 July 2025 (Latest Release)
0;32m[	info ] 2025/09/28 17:21:21 talkkonnect.go:67:0m Hardware Detected As Raspberry Pi 3 Model B Plus Rev 1.3
0;41m[ error ] 2025/09/28 17:21:21 client.go:335:0m Unable to Unmute  failed to execute "amixer set default unmute" (exit status 1)
0;32m[	info ] 2025/09/28 17:21:21 onevent.go:63:0m Connected to intercom-server Address 127.0.0.1:64738 on attempt 0 index [0]
0;32m[	info ] 2025/09/28 17:21:21 onevent.go:68:0m  Welcome to this server running Murmur.
0;32m[	info ] 2025/09/28 17:21:21 onevent.go:68:0m  Enjoy your stay!
0;32m[	info ] 2025/09/28 17:21:21 clientcommands.go:198:0m Joined Channel Name:  Intercom  ID	1
0;32m[	info ] 2025/09/28 17:21:21 client.go:362:0m Connected to Server Successfully
0;32m[	info ] 2025/09/28 17:21:21 client.go:470:0m Client Is Already Registered
0;96m[ debug ] 2025/09/28 17:21:21 clientcommands.go:246:0m  {0 0 Root <nil> map[]}
0;96m[ debug ] 2025/09/28 17:21:21 clientcommands.go:246:0m  {1 1 Intercom 0x1697000 map[4:0x16d63f0]}
?12l?25hBmH2J?1049l23;0;0t?1l?1006l?1015l?1002l?1000lpanic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
github.com/talkkonnect/talkkonnect.(*Talkkonnect).ClientStart(0x1758008)
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:566 +0x3734
github.com/talkkonnect/talkkonnect.Init({0x7e91c720, 0x21}, {0x7a69ac, 0x1})
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/client.go:235 +0x1654
main.main()
	/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/cmd/talkkonnect/main.go:78 +0x3a4
`
Any assistance would be greatly appreciated and thanks again for your efforts in maitaining talkkonnect!

Wainy

talkkonnect, 2025-10-01:
Please look in the accounts section I suspect that area of the talkonnect.xml config file.

talkkonnect, 2025-10-02:
Looks like the voice target is section is missing pls use the standard accounts section from the sample config files and try again. I think it is xml problem.